brilliantlabsAR/monocle-fpga #4: Cannot replay after turning off/on live graphics

josuah, 2023-02-15:
Here is the sequence that shows the problem:

```
import fpga

# enable piping from the camera to the memory ring buffer
fpga.write(0x1005, b'') # record on

# turn on live replay, we do see the camera output
fpga.write(0x3005, b'') # live on

# turn off live replay and back on, we still see the camera output
fpga.write(0x3004, b'') # live off
fpga.write(0x3005, b'') # live on

# turn off the recording, the image stays still
fpga.write(0x1004, b'') # record off

# now we can replay that recorded sequence multiple times if we wish to
fpga.write(0x3007, b'') # camera replay
fpga.write(0x3007, b'') # camera replay

# same as before, turn the live piping off and on again, we see the test pattern
fpga.write(0x3004, b'') # live off
fpga.write(0x3005, b'') # live on

# now replay does not want to show-up on camera replay, although we did just toggle live=off/on
fpga.write(0x3007, b'') # camera replay
```

I will try to look at this from up close.

sathibault, 2023-02-16:
In order to handle the switching between graphics display and camera display, it was necessary to reset the display controller.  Currently that happens whenever you send the 0x3004 or 0x4404.

josuah, 2023-02-16:
Thank you! I will move toward integrating it onto the firmware then.

closed 2023-02-17 by josuah